jsimpso81/WPIlibMathLabVIEW #28: How to use this awesome trajectory loader with WPILIB Java

mendax1234, 2021-12-08:
Do l need to install any venders?
Is the any API example in Java that l can refer to?

jsimpso81, 2021-12-09:
Hi.  I’m not certain what you are asking?   If you want to use the trajectory creation tools, they can make json files directly or convert cave files to json.  

If you install the trajectory library it will install the tools and create windows start menu items to launch them.  I think the only pretequisites are the frc tools ( driver station ). 

There is a separate repository for install files for the executables 

https://github.com/jsimpso81/FRC_LV_TrajLib_Utilities

let me know if that is what you were interested in

another team incorporated waypoint dragging.  I should have that merged soon.

mendax1234, 2021-12-09:
My meaning is that l use Java to write my robot's code. And if l want to use your tool to draw a trajectory and use it in autonomous . How should l do ? How to generate the trajectory in code and how to draw it?

Thanks for replying!

jsimpso81, 2021-12-11:
There are two steps in executing a trajectory on a robot:

1) Create the trajectory   -- create the Trajectory memory data structure (class in java).  a) The can be done off the robot using one of the available tools -- PathWeaver, PathFinder, PathPlanner, or the Trajectory creation tool that is part of this library.  The tools create a file that is copied to the robot.  (I think there is a deploy directory that is part of the Java robot project where these files are placed.)  At boot time the robot reads the file into a memory Trajectory class instance.  b) The robot executes code to create the trajectory from waypoints.

2) Execute the trajectory  -- run code on the robot to move the robot along the trajectory.  This uses the language of your robot -- Java in your case.  

This library, which is entirely in LabVIEW, can only help with item 1) -- creating the trajectory off the robot.  There are a couple different trajectory (or path) creation tools that run on Windows PCs.  They can create JSON files that are compatible with the WPILIB trajectory software.

The prerequisites for using just the these tools are:
  -- FRC game tools
  -- Windows 32 or 64 bit operating system.
Download the tools package from here, double click on the .NIPKG file to star the install.
https://github.com/jsimpso81/FRC_LV_TrajLib_Utilities/releases/tag/V2.04

The main link to this github repository is:
https://github.com/jsimpso81/FRC_LV_TrajLib_Utilities

As far as 2) executing the trajectory on your robot, this library can only help if your robot is programmed in LabVIEW.  For Java use the corresponding WPILIB functions.  In general the process is:
  -- Keep track of where your robot is.  Use the "Odometry udpate" routine for the type of robot you have.
  -- Sample the Trajectory -- This provides with where the robot should be at a particular time while executing the trajectory and how fast and in which direction your robot should be going.   From this a desired Chassis speed can be developed.  This can be used as a FEEDFORWARD to a closed loop controller.
  -- The difference between the Odometry routines indicate where your robot is and the sampled Trajectory State indicating where the robot should be is an ERROR value that can be fed to a closed loop controller.
  -- Use some form of closed loop controller to calculate the actual wheel speeds (and for swerve the direction).  There are a number of different controllers that can be used but the main ones are - Ramsete -- for differential drive robots, and Holonomic control for swerve drive robots.

For robots programmed in LabVIEW there are a number of sample robots -- which all can be simulated and driven around on a PC -- in this repository.  (Again this is of very minimal value if you are not programming your robot in LabVIEW.)
https://github.com/jsimpso81/FRC_LV_TrajLib_Util_and_Samp

Hopefully this can get you started.

Since this isn't really a bug (or issue) with this library, it might be better if you send me a message on Chief Delphi instead of here.  I'll probably wait a couple of days and then close this issue.

jsimpso81, 2021-12-14:
Closed.   This isn't really an issue or enhancement request.

closed 2021-12-14 by jsimpso81