jsimpso81/WPIlibMathLabVIEW #48: PathPlanner LoadPath fails with repeated error

MazeOfEncryption, 2022-03-25:
I successfully got swerve drive odometry and kinematics working using this library. Now, I was attempting to use PathPlanner.
For some reason, whenever I try to load a path file, I get this error, repeatedly (I have to spam click enter and hit `CTRL + .` to stop it).
[image]

I tried to determine what the cause of the error was by probing some of the subvi's (and temporarily setting them to non-reetrant).

Oddly enough, it seems that the file *is* being loaded and parsed successfully into an array of `PathPlannerWayPoints`. However, it seems that somewhere in the parsing of this trajectory, there's some sort of error. Any help would be appreciated!

MazeOfEncryption, 2022-03-25:
Just figured out that it was caused by holonomic mode being enabled in PathPlanner. Everything seems to work with it disabled. 
[image]

MazeOfEncryption, 2022-03-25:
Perhaps either add better error checking for this, or implement support for holonomic patholanner trajectories

MazeOfEncryption, 2022-03-27:
Interestingly, it seems to work with holonomic paths now? I didn't change anything...

Not sure what was going on.

jsimpso81, 2022-04-26:
This may be caused by a newer feature in pathplanner that hasn't been ported to LabVIEW yet.  I'll have to check on the latest pathplanner code for differences.  Thanks for posting the issue.

jsimpso81, 2022-04-27:
This library is based on the 2022.0.0 release.  I'm implementing the updates.   After that Is done, I'd like to try and duplicate the problem.  Can you send the JSON file that causes the error?

brandonzx3, 2022-05-11:
im getting this same error on a non holonomic path. not sure what to do

jsimpso81, 2022-05-11:
Can you send me the file you are trying to read. Also there are some changes to the pathfinder Labview library that aren’t in a published build.  Some relatively recent fixes were made to the Java version and I just finished putting these to Labview.  I’ll make a build for you to try.  However having the file will help me diagnose.   Thanks.

jsimpso81, 2022-05-13:
I think that this is fixed.   Some programmer....... left a debug pop up box in PathPlanner_Trajectory_CalculateMaxVel.vi
You can change it yourself, or a pre-release of the libraries is published with this fix...
https://github.com/jsimpso81/FRC_LV_TrajLib/releases/tag/V2.09Beta

Here is a project that I used for testing.  It runs on a PC..
[PathPlannerReadFileTest.zip](https://github.com/jsimpso81/FRC_LV_TrajLib/files/8691040/PathPlannerReadFileTest.zip)

Let me know if this works for you.

brandonzx3, 2022-05-21:
sorry for the delay. the pre-release fixed this issue. thank you for the help.

jsimpso81, 2022-05-22:
No troubles.   Glad its fixed.   Closed.

closed 2022-05-22 by jsimpso81