jsimpso81/WPIlibMathLabVIEW #123: [enhancement] way to turn csv file to path?

Not-AriStienfeld, 2023-01-19:
as far as I'm concerned, there isn't a block that turns csv files into paths. I feel like that would be pretty useful for coding pathfollowing code.

Not-AriStienfeld, 2023-01-19:
by path clusters I'm referring to what the yellow folder under "3rd party" uses

jsimpso81, 2023-01-19:
Thank you for your interest in the library.

There is a VI in this library that will read CSV or JSON files that are stored on the robot and create Trajectory data structure.  This allows Trajectories (Paths) that are created by PathWeaver or the LabVIEW trajectory creation utility to be copied to a robot, read by the robot (probably in the BEGIN.VI) the executed.  There is a sample robot that does this --- Sample 2 --  https://github.com/jsimpso81/WPIlibMathLabVIEW_Examples/tree/master/Robot_Sample_2_Diff_File_Traj_ODE

The VIs to read the JSON files are can be found here.    
[image]

[image]

The JSON read routine isn't as easy to find...  

[image]

[image]

[image]

Hope this helps.  If I interpretted your question incorrectly, please clarify.   If this solves your issue, please indicate that I can close the issue.  Thanks.

Not-AriStienfeld, 2023-01-19:
Hi, thanks for the quick response. I'm wondering if this works with the holonomic jsons created from the 2023 path planner application (the one on the windows store). I understand the new features won't work, but will the json itself work?

Not-AriStienfeld, 2023-01-19:
[image]
and I was mainly asking about how to use this block because I couldn't find a way to get the path as an array of clusters from the csv or json

Not-AriStienfeld, 2023-01-19:
[image]
while I'm at it, I also realized that with this odometry, my y is extremely accurate, but the x is all messed up. any time I stop moving in the x direction (the modules reset to their default "0" location) the x displacement goes to 0. I'm not sure why this is.

Not-AriStienfeld, 2023-01-19:
I fed the values back into the loop with a feedback node and it was worse...

Not-AriStienfeld, 2023-01-20:
I got it working by negating some values I can't close it from mobile, so you are good to close it

closed 2023-01-20 by Not-AriStienfeld