jsimpso81/WPIlibMathLabVIEW #160: Rasmete Controller Issue

WinstonHChou, 2023-05-16:
First of all, thank you for developing this WPILib Math LabVIEW Library! You've been a legend to many FRC LabVIEW teams.
I'd like to ask how to use Rasmete Controller in this library.
In my project of Diff Drive Trajectory following,  after making sure that my velocity controllers can keep up with the speed demand (for both left and right), I added Rasmete Controller to it. But it didn't work properly.
Suppose to be like:
[image]

The Result:

https://github.com/jsimpso81/WPIlibMathLabVIEW/assets/68310078/219c820d-6721-4a27-b051-ac337664eddc

WinstonHChou, 2023-05-16:
I've solved it after I debug my simulator with shift register and, critically, using ramsete controller w/ ext odmeter.

closed 2023-05-16 by WinstonHChou

jsimpso81, 2023-05-17:
I'm glad to hear that you fixed the issue.  Sorry I didn't see the issue until now.   If you have any suggestions for improving the documentation, please add a new issue with the suggested enhancements.