jsimpso81/WPIlibMathLabVIEW #162: DC motor Simulation

WinstonHChou, 2023-10-22:
I believe there a bug with Velocity_RPM's unit in DC motor simulation execute.
Here's a small expr I did that lead me to the doubt:
1. Create a 20ms loop to place a DC motor simulator execute vi.
2. give any motor power output
3. Using current value of Rotation minus the previous value, then divide it by / 20 ms
4. I find that the result approximately equals to the value of RPM

So, Shouldn't the unit of velocity be Rotation per sec?

jsimpso81, 2023-11-08:
Thanks.  I will look into this.  Is the problem with DCMotorSim_Execute.VI ?

Do you have a test program or screen shots that you can add here to help diagnose the issue?

Thanks !

jsimpso81, 2023-11-08:
I confirmed the bug, but it appears to be different.   It looks like the RPM calculation is correct.  The rotations calculation is off by a factor of 60.    (Take the current result of rotations and divide it by 60.)   Internally everything is in radians and radians/sec.   The conversion to RPM appears to be correct but the conversion to rotations is using the rad/sec to rpm conversion which is wrong.

Also if you are using the menu macro, there is a bug here too.  The simulated battery voltage VI should have a constant 12 as the "nominal voltage" input.  The macro uses the current simulated voltage which results in the voltage being driven down to 0 after a period of time.

jsimpso81, 2023-11-08:
[image]

WinstonHChou, 2023-11-08:
Thanks for identify the issues!!
How long does it take to update this correction?

jsimpso81, 2023-11-09:
Fixed.  Will be in the 3.08 release.  I'll probably publish it tomorrow.  (Also if you are using the battery simulation, the nominal voltage value is a constant.  The macro was wrong.  It is fixed now too.)    I tested it by simulating a CIM motor.  The RPM value matched what I expected.  I let it run for a minute and the rotations also matched what I expected -- not exact -- but fairly close.

closed 2023-11-09 by jsimpso81

jsimpso81, 2023-11-11:
In case you didn’t see it, a new release was published with the fix