jsimpso81/WPIlibMathLabVIEW #172: holonomic controller - feedforward

jsimpso81, 2023-11-25:
If -- desired heading - means robot orientation and not direction of travel, the feedforward calculations appear to be incorrect.   If the orientation is non-zero, this causes the feedforwards to be off.    

Just some clarification of my understanding:
- The current pose contains the current robot orientation as well as X, Y location
- The trajectory state (Pose Ref) contains the desire X, Y, direction of travel, and linear velocity
- The desired heading contains the desired robot orientation.

I think that the feedforwards should then be:
  - Xfeedforward = linear velocity * cosine( direction of travel - current actual orientation )
  - Yfeedforward = linear velocity * sine( direction of travel - current actual orientation )

Perhaps for a pure feedforward, the desired orientation could be used instead of actual.  (I think the actual values will provide better results.)

WinstonHChou, 2023-11-30:
Hi jsimpso81,
I noticed this too when I developed a swerve drive project for my team (FRC#8020) ,utilizing the provided examples of Swerve Drive. When not following orientation, the examples of either swerve or mecanum will give 0 to --desired heading--, which leads to incorrect feedforward values of Vx and Vy.
Thus, my implementation is when not following orientation, set vOmega output as 0, instead of setting --desired heading-- as 0.

Regards,
Winston

WinstonHChou, 2023-11-30:
Another problem I've got, when developing swerve drive trajectory following, is that swerve drive skew could lead to inproper follow orientation. Is there any known method to tackle the skew problem?

FYI, this is the discussion thread of swerve skew issue on chief delphi.
https://www.chiefdelphi.com/t/whitepaper-swerve-drive-skew-and-second-order-kinematics/416964/8

WinstonHChou, 2023-12-03:
> If -- desired heading - means robot orientation and not direction of travel, the feedforward calculations appear to be incorrect. If the orientation is non-zero, this causes the feedforwards to be [...]

Hi @jsimpso81 ,
I happen to notice that in the previous version,
**X_ff = linear vel * cos(angle_ref) = linear vel * cos(Pose_Ref)**
**Y_ff = linear vel * sin(angle_ref) = linear vel * sin(Pose_Ref)**
if you set [angle_ref = Pose_Ref], then the both FF would be correct.
It means that Pose_Ref equals to path's direction of travel.

Thus, I think that
**X_ff =linear vel * cos(Pose_Ref)**
**Y_ff = linear vel * sin(Pose_Ref)**
should be enough.
Noticed that X_ff, Y_ff is **with respect to Field-Relative Coordinates**, 
you don't have minus current acutual orientation again when you have From_Field-Relative-Speeds.vi at the end of the calculation.

jsimpso81, 2024-03-16:
I think this is ready to close ???

WinstonHChou, 2024-03-16:
Yup

closed 2025-01-12 by jsimpso81