jsimpso81/WPIlibMathLabVIEW #174: Swerve Skew Issue

WinstonHChou, 2023-12-03:
A problem I've got, when developing swerve drive, is that swerve drive skew leads to undesired route moving. Is there any known method to tackle the skew problem?

FYI, this is the discussion thread of swerve skew issue on chief delphi.
https://www.chiefdelphi.com/t/whitepaper-swerve-drive-skew-and-second-order-kinematics/416964/8

WinstonHChou, 2023-12-04:
> Take a look at the swerve drive odometry example. It has a crude implementation of logic to try and minimize swerve skew. Good tuning of the swerve module position control is also essential. In the simulation and example, I show three different types of control --- PID with characterized error, profiled PID, and Position control. For the simulation position control seemed to work the best. However I didn't spend much time tuning.

Is the example within the release library package? or should I git clone the repository?

jsimpso81, 2023-12-05:
The release has 2 install files.  One contains the library.  The other contains the examples.  ( There are actually 4 files in the release - a set for LabVIEW 2020 and another set for LabVIEW 2023.  Make certain to choose the files that match your version of LabVIEW.  The examples can then be located through LabVIEW's "Find examples" menu.

jsimpso81, 2024-03-16:
Is this issue ready to be closed??

WinstonHChou, 2024-03-16:
Nah still cannot resolve this issue. Is there any good way to tune steering controller? PID is not a good option.