jsimpso81/WPIlibMathLabVIEW #181: Inquiry About Continued Maintenance for 2025 FRC Season

WinstonHChou, 2025-01-06:
Hi @jsimpso81 ,

First, thank you for creating and maintaining this valuable repository—it has been a tremendous help to LabVIEW teams and the broader FRC community.

As the 2025 FRC season approaches, I wanted to check in and ask if you plan to continue maintaining this repo. I completely understand that this is a volunteer-driven effort, and there's no obligation or pressure to do so. Your contributions have already made a significant impact.

For context, my team has decided to stick with LabVIEW this year, and we rely heavily on the tools and features provided here. If you are open to it, I'd be happy to contribute by adding new features or addressing issues through PRs to help lighten the load.

Let me know your thoughts, and thanks again for all your hard work!

Best regards,
Winston

jsimpso81, 2025-01-08:
Short answer - Yes.

While I have not tested yet for 2025, since LV 2023 is being used, the install files for the libraries (Math, Photonvision, Choreo, Path Planner) should install.   Changes to Photonvision and maybe others may not work until that review and update is complete.  (In other words use the 2024 photonvision for now...)

Here is a list of to do items for the first release this year.
  - Math lilb:
  - - april tag positions for 2025
  - - field picture for 2025
  - - efficiency improvements for a number of VI - filter, vision things
  - PhotonVision
  - - update for 2025 packet format if it changed
  - PathPlanner
  - - update for 2025 file format if it changed

Enhancement suggestions and suggestions for priorities are welcome.

WinstonHChou, 2025-01-09:
I was thinking to add Kraken X60 data to motor select
[image]
the image above is the theoretical values that my team uses

jsimpso81, 2025-01-11:
Yes.   I was going to change the motor model  to be a polymorphic VI and add add the new motors.  If you have done some already, I'd be happy to include.

WinstonHChou, 2025-01-11:
I’m really interested in contributing to this project and would like to get started by compiling the library on my own. However, I couldn’t find detailed instructions on how to do so.
Could you kindly provide guidance or documentation on how to set up and compile the wpimath LabVIEW library? Any tips regarding dependencies, build tools, or specific steps would be greatly appreciated.

Note: I've made some simple SysIdRoutine vis and wpilog vis to allow LabVIEW users to use 2024 & 2025 sysid tool
It would be nice to integrate the feature into the library

jsimpso81, 2025-01-12:
I'll send you instructions on building the library.   Some parts are simple.  Other parts - documentation - are not so simple (but could be improved...)

I'll close this issue for now and reach out directly.

closed 2025-01-12 by jsimpso81

WinstonHChou, 2025-01-18:
Polymorphic VIs may not be the most feasible solution for motor models, as it would be challenging to build a case structure to dynamically select the appropriate model using an enum variable.

jsimpso81, 2025-01-20:
I found that out as I was going through the users of this VI.    So I kept both -- the polymorphic VI, and the one driven by the ENUM input.    The VI isn't in a menu but you can add it manually:   .....\vi.lib\Rock Robotics\WPI\ThirdParty\JAS_Junk\TrajLib\DCMotor_PickMotorSelector.vi

I'll add an issue to get this re-added to the menus.

jsimpso81, 2025-01-20:
Something else you might be interested in -- I've been suggesting to the team I mentor that they create systems that can be easily simulated on a computer before trying on the robot.   You can find the code the their latest stuff here.

https://github.com/FRobotics/2024RobotCode/tree/main/Barcie

WinstonHChou, 2025-01-20:
Thanks for sharing! I'll definitely check it out. I'm especially intrigued by the potential of using simulation programs to test logic before deploying it to the robot.

Inspired by your labview examples, I developed our own system, which now supports swerve drive simulation. After a year of development, I’ve decided to make it open-source:

https://github.com/C-K-Robotics/FRC8020-Swerve-Drive-Project

WinstonHChou, 2025-01-25:
> I'll send you instructions on building the library. Some parts are simple. Other parts - documentation - are not so simple (but could be improved...)
> 
> I'll close this issue for now and reach out directly.

Hi @jsimpso81 

Thanks for the update! Please feel free to reach out to me directly at winston.h.chou@gmail.com for the instructions.